fossasia/pslab-app #1947: Oscilloscope main screen UI changes

neel1998, 2019-08-17:
**Actual Behaviour**
Make following changes to Oscilloscope main screen UI
1. Add radion button for pslab mic and built in mic
2. Align spinners
3. add spacing between different sections
4. reduce the width of the left control panel

**Would you like to work on the issue?**
Yes

rajput2122, 2019-08-21:
sir, I want to fix this issue. can you allow me to do the changes and contribute

neel1998, 2019-08-21:
@rajput2122 sure. Go ahead. Ask any doubts related to code base on the bitter channel. Once the changes are done make a PR in specified format. Cheers!

rajput2122, 2019-08-21:
thank you sir :)

mariobehling, 2019-08-21:
@rajput2122 It it is difficult to dive into what we discussed in detail in meetings recently. Please choose an easier issue first. @neel1998 Please finalize the work you have stated in your scrum with this issue and make a PR as discussed in the last meeting. We dont have time to assign issues we discussed in detail in our meetings to others. Thanks.

neel1998, 2019-08-21:
@mariobehling sure will make a PR asap

rajput2122, 2019-08-21:
@mariobehling ok sir thanks for guidance

PR #1955 (2019-08-27): fix: osc main screen ui changes
Fixes #1947 

**Changes**: 
1. Radio buttons for mic added
2. Margins between components added
3. Some text boxes aligned
4. Left control panel width reduces

**Screenshot/s for the changes**:
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR  [... cut]
closed 2019-08-27 by CloudyPadmal