fossasia/pslab-app #1964: Update readme file in apk branch

CloudyPadmal, 2019-08-28:
**Actual Behaviour**

APK branch readme is outdated and it's not synced with development branch docs.

**Expected Behaviour**

This readme file needs to be updated with similar content to neurolab-android repo.
- QR codes for both Fdroid and Playstore links
- Redirect links
etc

**Steps to reproduce it**

Check out https://github.com/fossasia/pslab-android/blob/apk/README.md

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Beginner issue

XDivergent, 2019-09-02:
Can I work on this Issue?

neel1998, 2019-09-02:
@XDivergent sure go ahead.

XDivergent, 2019-09-02:
> This readme file needs to be updated with similar content to neurolab-android repo.

Is that mean I need to delete most of the current contents and add only content similar to **neurolab-android repo** readme?

CloudyPadmal, 2019-09-10:
@XDivergent yes correct. You'll have to delete those outdated content and make it similar to what neurolab-android repo has.

XDivergent, 2019-09-15:
@CloudyPadmal Sir, Can I submitted PR after 20 sept. I am having exams till then.

XDivergent, 2019-09-19:
[image]
@CloudyPadmal @neel1998

XDivergent, 2019-09-23:
@fossasia/pslab-admins

CloudyPadmal, 2019-09-23:
Hello @XDivergent looks good. Can make a PR?

tusharr-98, 2019-10-07:
@CloudyPadmal is this issue still open?

refactor-droidyy, 2019-10-25:
Hello @CloudyPadmal can I work on this?

closed 2019-10-27 by CloudyPadmal