fossasia/pslab-app #1967: Add degree symbol to wavegenerator

CloudyPadmal, 2019-08-29:
**Actual Behaviour**

Current UI has `deg` to indicate degrees

**Expected Behaviour**

Use correct symbol instead

**Steps to reproduce it**

Check out Wave generator UI

**LogCat for the issue**

> Not available

**Screenshots of the issue**

> Not available

**Would you like to work on the issue?**

Beginner issue

XDivergent, 2019-08-31:
Can I work on this issue?

XDivergent, 2019-08-31:
[image]

what about space between value and symbol? shall I remove it or not?

CloudyPadmal, 2019-09-01:
Hello @XDivergent , Let's remove the space. Once done, make a PR with the changes. Thanks.

XDivergent, 2019-09-01:
@CloudyPadmal I remove the space between value and symbol but it would remove spaces for other symbols also like `Hz` and `%`.
So do you want to keep the spaces for other symbols or remove space for every symbol?
[image]

CloudyPadmal, 2019-09-01:
@XDivergent make changes only for the degree symbol

PR #1972 (2019-09-02): fix: deg text to symbol.
Fixes #1967 

**Changes**: [deg was written as text changed to its symbol (\u00b0) ]

**Screenshot/s for the changes**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [ ] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [ ] My code does not contain an [... cut]
commit 8dd2c0f8 (2019-09-02): Fixes #1967 deg text to its symbol (#1972)
closed 2019-09-02 by CloudyPadmal