fossasia/pslab-app #1973: Improving the UI of generate config file activity

ho-dor, 2019-09-09:
**Actual Behaviour**

UI of generate config file doesn't look ordered. Lot of empty space on screen

**Expected Behaviour**

Checkboxes should be assigned, more screen space should be covered

**Steps to reproduce it**

Open Generate config file from nav bar.

**LogCat for the issue**

N/A

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**
Yes

kshitiz-kumar, 2019-09-29:
anyone working on this

kshitiz-kumar, 2019-09-29:
If not I can work.

kshitiz-kumar, 2019-09-30:
@neel1998  I would like to work on this.Please assign this to me.

PR #1977 (2019-10-18): Improved UI Of Config Activity
Fixes #1973

**Changes**: The Config Activity UI was unordered, so changes were made in the config layout by adding a number of views.

**Screenshot/s for the changes**: 
<image src = "https://user-images.githubusercontent.com/38812752/65950873-8fab4080-e45c-11e9-879e-fbd7b06c0d89.jpeg" width=250>

<image src ="https://user-images.githubusercontent.com/38812752/65950876-90dc6d80-e45c-11e9-8c99-0f199cd938b9.jpeg" width=250>

[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from ` [... cut]
closed 2019-10-18 by CloudyPadmal