fossasia/pslab-app #2000: 'Not Connected' toast message repeatedly keeps popping up in Dust Sensor

reckoner1429, 2019-11-09:
**Actual Behaviour**

When dust sensor is not connected, the **not connected** toast message keeps popping up repeatedly.

**Expected Behaviour**

The **not connected** message must not keep flashing  before the screen just like it is in other sensors where the message is displayed only once

**Steps to reproduce it**

- Do not connected the USB device and then open the Dust Sensor

**LogCat for the issue**

>N/A

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

shreyansjain012, 2019-11-12:
Hi. Is this issue assigned to someone else? I would live to work on it

adityastic, 2019-11-12:
@shreyansjain012 we might need input from @reckoner1429 atm

reckoner1429, 2019-11-12:
@shreyansjain012 and @adityastic , I also want this issue to be assigned to me, however I have not received any word from the repository maintainers. 
PS - This is my first attempt at open source contributions so i am not very well aware about how long will i have to wait to receive a word from the maintainers

neel1998, 2019-11-12:
@reckoner1429 since you have raised the issue and also mentioned you would like to work on it there is no need for you to wait for the maintainers to assign the issue to you. You can start working on it and make the PR. Please refer to the readme for contributors and follow the best practices . Cheers.

shreyansjain012, 2019-11-12:
@reckoner1429 @adityastic I am new to open source I think I have the code ready which is working fine but I don't know how to proceed from here? Should I wait for someone to assign it or just submit a pull request?

neel1998, 2019-11-12:
@shreyansjain012 it's great that you have solved this issue but since the author of the issue already mention that he would like to work on it is not good practice to file a PR until the author says he is not gonna work on it anymore. In this case @reckoner1429 if is not going to work on this issue anymore you can file a PR

shreyansjain012, 2019-11-12:
@neel1998 Okay. Thanks.

reckoner1429, 2019-11-12:
@neel1998 Thanks a lot for the directions! I am now working on this issue.

PR #2002 (2019-11-12): Fixes #2000 Fixed the repeated display of 'Not Connected' Toast messa…
…ge in DustSensor

Fixes #2000

**Changes**: 
Fixed the repeated display of 'Not Connected' Toast message in DustSensor.
Made Changes to  `onCreateView()` and `visualizeData()` in `DustSensorDataFragment` 

**Screenshot/s for the changes**: 
[image]

**Checklist**:
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in thi [... cut]
commit 33c4c7c6 (2019-11-12): Merge pull request #2002 from reckoner1429/development

Fixes #2000 Fixed the repeated display of 'Not Connected' Toast messa…
commit a1f8f598 (2019-11-12): Fixes #2000 Fixed the repeated display of 'Not Connected' Toast message in DustSensor
closed 2019-11-12 by neel1998