fossasia/pslab-app #2004: Menu button does not works

shreyansjain012, 2019-11-14:
**Actual Behaviour**
When you click on **Menu** button nothing happens

**Expected Behaviour**
Not known

**Steps to reproduce it**
Do the following
1. Go to Navigation menu
2. Click on **Connect Device**
3. Click on **What is PSLab Device?**
4. Click on **Menu** Button on the top right corner
 

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

dheerajkotwani, 2019-11-16:
@CloudyPadmal can you tell the expected behaviour on pressing the menu button there?

dheerajkotwani, 2019-11-16:
@shreyansjain012 this page is the webpage of site "pslab.io" in mobile view. Pressing this button in mobile web view open navigation drawer of the webpage.

amit6663, 2020-01-02:
can i work on this issue @shreyansjain012 ?

shreyansjain012, 2020-01-09:
Yes please go ahead @amit6663

PR #2069 (2020-01-17): Fixes #2004 Fixes the menu button in the webview of the website
Fixes #2004 

**Changes**: Enabled JavaScript so that the menu button could work properly and enabled DOM Storage so that the data of current web session is stored

**Screenshot/s for the changes**: 
[image]

**Checklist**:
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd> [... cut]
commit 2082f7cf (2020-01-17): Fixes #2004 Fixes the menu button in the webview of the website (#2069)
closed 2020-01-17 by CloudyPadmal