fossasia/pslab-app #2008: App crash on Creating Config File

dheerajkotwani, 2019-11-16:
**Actual Behaviour**
When we try to Create Config File the app crashes.

**Expected Behaviour**
Config File should be created

**Steps to reproduce it**
1.Open Generate Config File from Navigation Drawer
2.Give some value to time interval
3.Click on Create Config File

**LogCat for the issue**
`E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.pslab, PID: 20018
    java.lang.NoSuchMethodError: No static method join(Ljava/lang/CharSequence;Ljava/lang/Iterable;)Ljava/lang/String; in class Ljava/lang/String; or its super classes (declaration of 'java.lang.String' appears in /system/framework/core-oj.jar)
        at io.pslab.activity.CreateConfigActivity.createConfigFile(CreateConfigActivity.java:187)
        at io.pslab.activity.CreateConfigActivity.access$800(CreateConfigActivity.java:31)
        at io.pslab.activity.CreateConfigActivity$3.onClick(CreateConfigActivity.java:111)
        at android.view.View.performClick(View.java:5619)
        at android.view.View$PerformClick.run(View.java:22298)
        at android.os.Handler.handleCallback(Handler.java:754)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:165)
        at android.app.ActivityThread.main(ActivityThread.java:6375)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:912)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:802)`

**Would you like to work on the issue?**
Yes, but please guide me for the actual behaviour for this

kshitiz-kumar, 2019-11-19:
Can share the ss or gif  because everything is working fine for me.

amit6663, 2020-01-02:
i want to work on this issue. @dheerajkotwani , @asitava1998

ho-dor, 2020-01-17:
Unable to produce this crash. @CloudyPadmal can you verify or this can be closed.

Low012, 2020-02-04:
It crashes on devices with API level 25 and below.

PR #2103 (2020-04-23): Fixes #2008 Fixes crash which occurred in Create Config File
Fixes #2008

**Changes**: Use join method of Apache StringUtils instead of the join method of the Android String class. The join method was added to the String class of Android in API level 26, but the minimum API level of the app is 23.

**Checklist**:
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines or extra spaces
- [x] I hav [... cut]
commit d331dfa0 (2020-04-23): Merge pull request #2103 from Low012/bugfix/2008

Fixes #2008 Fixes crash which occurred in Create Config File
commit f2784c01 (2020-04-23): Fixes #2008 Fixes crash which occurred in Create Config File on devices with API level < 26
closed 2020-04-23 by cweitat