fossasia/pslab-app #2009: 'Show Logged Data' menu of Dust Sensor showing logged data from other sensors as well

reckoner1429, 2019-11-20:
**Actual Behaviour**
On Opening the *Show Logged Data* Menu of Dust Sensor, the saved logs of all the sensors are being displayed

**Expected Behaviour**
On Opening the *Show Logged Data* Menu of Dust Sensor, only the logs that belong to the Dust Sensor must be displayed.

**Steps to reproduce it**
1. Click on the Dust Sensor Option from the Main Menu
2. Click on the 'Overflow Menu' button.
3. Click on 'Show Logged Data' Option.

**LogCat for the issue**
`Not Applicable`

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes, definitely!

amit6663, 2020-01-02:
@asitava1998 can i work on this issue?

reckoner1429, 2020-01-02:
@amit6663 This issue already has a fix. [#2023 ](https://github.com/fossasia/pslab-android/pull/2023)

PR #2023 (2020-01-15): Fixes #2009 Fixed the issue of 'Show Logged Data' menu option of Dust…
…Sensor showing data from all sensors

Fixes #2009

**Changes**: Fixed the issue of 'Show Logged Data' menu option showing logs of all other sensors

**Screenshot/s for the changes**:
[image]

**Checklist**:
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] M [... cut]
commit 24c2e5e5 (2020-01-15): Fixes #2009 Fixed the issue of 'Show Logged Data' menu option of DustSensor showing data from all sensors (#2023)
closed 2020-01-15 by mariobehling