fossasia/pslab-app #2017: App crashes

chamodperera, 2019-12-09:
### App crashes when recording oscilloscope

- **I opened oscilloscope in pslab**

[image]

- **And I selected In-Built Mic**

[image]

- **when I try to record it the app crashes**

[image]

reckoner1429, 2019-12-12:
Perhaps it could be some device specific issue, I tried to reproduce the bug but didn't get any results, the built in mic recording is working fine for me. Perhaps uploading the logcat of the crash would give more insights.

ho-dor, 2020-01-17:
Same as #2081. @CloudyPadmal please verify or we can close this.

CloudyPadmal, 2020-01-18:
Similar issue raised #2032

closed 2020-01-18 by CloudyPadmal