fossasia/pslab-app #2019: Crashing of the App while using the Logged Data

aryanguls, 2019-12-09:
**Actual Behaviour**
On pressing the icon next to the delete option in the logged data(images provided below) the app crashes followed by complete blacking out of the phone screen.

**Steps to reproduce it**
1) Use any of the gadgets present in PSLab such as the oscilloscope or the lux meter.
2) Record the data observed by the instrument.
3) Go to the drop down menu and select the logged data option.
4) Press the icon to the left of the bin/delete icon.
5) The app crashes on doing so.

**Screenshots of the issue**
[image]
[image]
[image]
[image]

**Would you like to work on the issue?**
Yes if given the option I would like to work on the problem.

amit6663, 2020-01-02:
can i work on this issue? @aryanguls

ho-dor, 2020-01-17:
Unable to produce this crash @CloudyPadmal please verify or else we can close this

CloudyPadmal, 2020-01-18:
Hello @aryanguls , can you please report the issue when the dialog shows when the app is crashed? It would help us see what's causing the issue. Thanks.

cweitat, 2020-04-23:
No response, assume is working. Reopen if problem still persists

closed 2020-04-23 by cweitat