fossasia/pslab-app #2032: Oscilloscope: App crashes as soon as record button is clicked while using built-in mic.

hyPnOtiC-d00g, 2019-12-15:
**Actual Behaviour**
Oscilloscope: The app crashes as soon as the record button is clicked while using the built-in mic.

**Expected Behaviour**
The shouldn't crash when the record button is clicked, and I should be able to obtain the .csv files after recording.

**Steps to reproduce it**
• Open the app & click on the Oscilloscope.
• Switch to built-in mic & click on record button.
• The app crashes.

**Screenshots of the issue**
None. 

**Would you like to work on the issue?**
No.

ho-dor, 2020-01-17:
Can you share the logcat please @hyPnOtiC-d00g

ho-dor, 2020-01-17:
I am unable to reproduce this error @CloudyPadmal this is same as #2081. Working fine on my device.

CloudyPadmal, 2020-01-18:
I see. @ho-dor may be it's some permission issue in microphone? :thinking: 

Sadly we can't expect community to share logcats. Unless they "Report" it, we could see it from Google Play console. (Couldn't see any as noone has "Report"ed it yet) Let's ask to report it.

dheerajkotwani, 2020-01-22:
The Issue is there.
Clicking on use in-built mic and then start recording.
**Permission for mic given to the app**

**LogCat**
`E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #4
    Process: io.pslab, PID: 8223
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:318)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
        at java.util.concurrent.FutureTask.run(FutureTask.java:242)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:243)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:760)
     Caused by: java.lang.NoSuchMethodError: No static method join(Ljava/lang/CharSequence;[Ljava/lang/CharSequence;)Ljava/lang/String; in class Ljava/lang/String; or its super classes (declaration of 'java.lang.String' appears in /system/framework/core-oj.jar)
        at io.pslab.activity.OscilloscopeActivity$CaptureTask.doInBackground(OscilloscopeActivity.java:1120)
        at io.pslab.activity.OscilloscopeActivity$CaptureTask.doInBackground(OscilloscopeActivity.java:960)
        at android.os.AsyncTask$2.call(AsyncTask.java:304)
        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:243) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607) 
        at java.lang.Thread.run(Thread.java:760) `

Low012, 2020-01-28:
String.join was added with API level 26. The min API level of the app is set to 23. I can reproduce the issue on a device with Android 7.1 (API level 25).

I can work on this issue.

PR #2102 (2020-04-23): Fixes #2032 Fixes crash which occurred in the oscilloscope
Fixes #2032

**Changes**: Use join method of Apache StringUtils instead of the join method of the Android String class. The join method was added to the String class of Android in API level 26, but the minimum API level of the app is 23.

**Checklist**:
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines or extra spaces
- [x] I have  [... cut]
commit 704244d7 (2020-04-23): Fixes #2032 Fixes crash which occurred in the oscilloscope on devices with API level < 26
commit b35835bd (2020-04-23): Merge pull request #2102 from Low012/bugfix/2032

Fixes #2032 Fixes crash which occurred in the oscilloscope
closed 2020-04-23 by cweitat