fossasia/pslab-app #2048: Add a language - Polish

rradzzio, 2019-12-29:
**Actual Behaviour**

The app does not have a Polish language

**Expected Behaviour**

Full support for one additional language - Polish

**Would you like to work on the issue?**
Yes

amit6663, 2020-01-18:
can i work on this issue? @rradzzio , @ho-dor

ho-dor, 2020-01-18:
Sure

CloudyPadmal, 2020-01-18:
Hello @amit6663 , unless you're a native Polish speaker, or know the language very well, I would suggest not to attempt different language translations from Google translate

amit6663, 2020-01-19:
> Hello @amit6663 , unless you're a native Polish speaker, or know the language very well, I would strongly suggest not to attempt different language translations from Google translate.
> 
> Also I see @rradzzio is from Poland and we better let him continue with a PR.

ok

PR #2049 (2020-02-04): fixes #2048 - add a Polish language
Fixes #2048 

**Changes**: Added a Polish language to strings.xml 

**Screenshot/s for the changes**: 
[image]
[image]
commit 5b2ce266 (2020-02-04): fixes #2048 - add a Polish language (#2049)
closed 2020-02-04 by mariobehling