fossasia/pslab-app #2056: Record option not working (Wave Generator)

NishithP2004, 2020-01-01:
**Actual Behaviour**
The record option present in the wave generator does not record the wave.
The files obtained on recording are empty and are only 35 Bytes in size .

**Expected Behaviour**
On pressing the save option,the app should record the wave and save it as .CSV file.

**Steps to reproduce it**
* Open PSLab Android App
* Select Wave Generator option
* Click the Save icon in the top right corner of the app
* Produce a wave

**LogCat for the issue**

**Screenshots of the issue**

**Would you like to work on the issue?**

rakesh-1601, 2020-01-09:
i would like to work on this issue.

NishithP2004, 2020-01-09:
Go ahead :)

ankit2019j, 2020-01-10:
can i work on this issue?

NishithP2004, 2020-01-10:
Sure

Ananyaas, 2020-09-26:
hey ! if this issue is still open ,may I work on this issue?

NishithP2004, 2020-09-28:
> hey ! if this issue is still open ,may I work on this issue?

Sorry for the late reply, You may proceed with this issue. :)

PR #2215 (2021-06-04): Fix: Wave Generator saving empty csv file
Fixes #2056
Fixes #2045 

**Changes**: No more checking for connected device as wave generator works with device speakers

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not  [... cut]
closed 2021-06-04 by CloudyPadmal