fossasia/pslab-app #2063: Misaligned buttons

aa200400, 2020-01-07:
**Describe the bug**

In the "Generate Config File", the buttons for instrument and time interval are misaligned and are partially cut out.

**To Reproduce**

1. Open the tabs to the right
2. Open "Generate Config File" screen
**Expected behavior**

The buttons should not be misaligned and out of place.
**Screenshots**

[image]

**Smartphone Info:**

|               | |
|---------------|-|
|Device         | OnePlus 6|
|Android Version|   10.0 |

**Additional context**

**Would you like to work on the issue?**

- [ ] Yes
- [x] No
- Other:

rakesh-1601, 2020-01-09:
After fix there is still some misalignment . can i work on this issue ?

angmas1, 2020-01-09:
Sorry, fixed with recent commit.

rakesh-1601, 2020-01-10:
Okay :)

cweitat, 2020-04-23:
Fixed by another PR

closed 2020-04-23 by cweitat