fossasia/pslab-app #2065: Dark Color Scheme

Sid911, 2020-01-09:
## Objective
Dark color scheme or **Dark mode** for the pslab application

_**Please wait for more description**_
## Goals

- [ ] Have a perfect dark scheme setup with accent color of red
- [ ] User's Ability to change between light mode and dark mode

**Would you like to work on the issue?**

Yes I am working on this issue :)

Sid911, 2020-01-17:
[image]
[image]
[image]

I know I might have flipped the colors of the action bar and the background but I was wondering weather to keep the accent in the card or the action bar? A good dark theme would have dark cards but action bar can be like of phimp.me with red accents?

(there are a few issue with black images, I am currently working on fixing them)

Sid911, 2020-01-17:
@mariobehling @fcartegnie @cweitat @adityastic It would be nice to have opnion .. on the color scheme 😇

Sid911, 2020-01-17:
Also the dark theme switches with system settings..

hoangdatngo, 2020-01-17:
@Sid911 LGTM

Sid911, 2020-01-17:
@hoangdatngo Thanks 😇

mariobehling, 2020-01-17:
Currently out of scope as we have more basic things on the list. Therefore closing it.

closed 2020-01-17 by mariobehling