fossasia/pslab-app #2070: pslab app crashes when selecting oscilloscope

corsch, 2020-01-12:
HUAWEI MediaPad M5 Lite 

Model BAH2-W19
Build BAH2-W19 8.0.0.238(OCEC432)
Android Version 8.0.0

App Version 2.0.20
Play Store version

**Actual Behaviour**

pslab app crashes when selecting the Oscillosope

**Expected Behaviour**

Show the Oscilloscope

**Steps to reproduce it**

1. open pslab app
2. Select oscilloscope 
3. app crashes

**LogCat for the issue**
--------- beginning of crash
01-12 17:42:39.467  3003  3003 E AndroidRuntime: FATAL EXCEPTION: main
01-12 17:42:39.467  3003  3003 E AndroidRuntime: Process: io.pslab, PID: 3003
01-12 17:42:39.467  3003  3003 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{io.pslab/io.pslab.activity.OscilloscopeActivity}: java.lang.IllegalStateException: Required view 'img_arrow_oscilloscope' with ID 2131296672 for field 'arrowUpDown' was not found. If this view is optional add '@Nullable' (fields) or '@Optional' (methods) annotation.
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3303)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3411)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.app.ActivityThread.-wrap12(Unknown Source:0)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1994)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:108)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:166)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:7529)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:245)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:921)
01-12 17:42:39.467  3003  3003 E AndroidRuntime: Caused by: java.lang.IllegalStateException: Required view 'img_arrow_oscilloscope' with ID 2131296672 for field 'arrowUpDown' was not found. If this view is optional add '@Nullable' (fields) or '@Optional' (methods) annotation.
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at butterknife.internal.Utils.findRequiredView(Utils.java:92)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at butterknife.internal.Utils.findRequiredViewAsType(Utils.java:104)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at io.pslab.activity.OscilloscopeActivity_ViewBinding.<init>(OscilloscopeActivity_ViewBinding.java:50)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at java.lang.reflect.Constructor.newInstance0(Native Method)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at java.lang.reflect.Constructor.newInstance(Constructor.java:334)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at butterknife.ButterKnife.createBinding(ButterKnife.java:199)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at butterknife.ButterKnife.bind(ButterKnife.java:124)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at io.pslab.activity.OscilloscopeActivity.onCreate(OscilloscopeActivity.java:220)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.app.Activity.performCreate(Activity.java:7383)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1218)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3256)
01-12 17:42:39.467  3003  3003 E AndroidRuntime:        ... 9 more
--------- beginning of system

dheerajkotwani, 2020-01-22:
@corsch Working fine for me.
[image]

corsch, 2020-01-22:
My Nexus 5x works fine too. ;)

BUT the app is still crashing on my HUAWEI MediaPad M5 Lite.

PR #2120 (2020-07-22): Fixes #2070 Add missing elements to Oscilloscope layout for tablets
Fixes #2070 

**Changes**: The layout for phones and tablets was out of sync. The tablet version was missing several elements which were present in the phone version and which the Activity relied on. I added the missing elements to the layout for tablet devices. I have added values to dimens.xml for tablet devices.

**Screenshot/s for the changes**: 
[image]

**Checklist**:
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource f [... cut]
commit f94ca699 (2020-07-22): Fixes #2070 Add missing elements to Oscilloscope layout for tablets (#2120)

* Updated strings.xml

* Fixes #2070 Add missing elements to Oscilloscope layout for tablet devices

* Fixes #2070 Remove outdated and superfluous layout for tablets
closed 2020-07-22 by CloudyPadmal