fossasia/pslab-app #2071: Add limit in input value

juniorjainsahab, 2020-01-13:
**Actual Behaviour**
In Robotic Arm Screen, the User can input as large value as they can in servo sections, Which causes the value out of its view bounds.

**Expected Behaviour**
Users should not be allowed to input more than 360. 

**Steps to reproduce it**

- Open the Application.
- Navigate to a Robotic-arm screen.
- Enter large values like "338439483" in any servo section.
- You will found that the values cross the view bounds.

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**
Yes, I would like to work on the issue.

991rajat, 2020-01-13:
Yes, I would like to work on the issue.

juniorjainsahab, 2020-01-13:
@991rajat I have raised the issue and i am waiting for maintainers to approve this so I can start working on this. I have already stated that I am interested to work on this issue. @CloudyPadmal  @cweitat could you please do something in this issue?

991rajat, 2020-01-13:
@juniorjainsahab Sorry I didn't know that. I thought you opened the issue publicly. I will close my PR. Sorry for this

juniorjainsahab, 2020-01-14:
@991rajat make it close once maintainers approve this issue I will start work on this.

Sid911, 2020-01-14:
what if the user wants to make the servo do multiple loops 😕... it may be that .. the robotic arm's specific servo is being used for different purpose

juniorjainsahab, 2020-01-14:
hey @Sid911 
As you can see the blue circular button to set value in the servo section it is limited to 360 as you can also see it in the image. Hence, It should be the same when you entering the value manually.

[image]

PR #2073 (2020-01-15): feat: Added input limit
Fixes #2071 
Added Input Limit

**Changes**: Added new MinMaxFilter Class

**Screenshots for the changes**: 

[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [ ] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra line [... cut]
closed 2020-01-15 by mariobehling