fossasia/pslab-app #2083: bug: About page keeps Shifting theme which leads to crashes in android 9 and 10

Sid911, 2020-01-19:
**Actual Behaviour**

Currently the application's About page randomly keeps shifting themes in android 9 and 10 *(maybe also 8 but I couldn't verify this)* when system theme is set to dark and application is switched and also causes crashes in the case of my android 10 testings. 

[image]
[image]
[image]

**Expected Behaviour**

The application theme shouldn't switch in fly it should remain the same and should never cause crashes

**Steps to reproduce it**
- Install pslab into a Android 9 or 10 phone with dark theme on
- go to about page ( you should see the dark theme of the about page)
- now switch application and come back to pslab

**Resources of the issue**
in android 9
[image]

in Android 10
[image]

**(it doesn't crash everytime but it does in 1/4 scenario in my avd testings)**

**Would you like to work on the issue?**

Yes I would like to work on the issue

marcnause, 2024-11-09:
Fixed in https://github.com/fossasia/pslab-android/pull/2084

marcnause, 2024-11-10:
Wrongly closed. PR wart closed, not merged. Investigate if issue still exists.