fossasia/pslab-app #2093: bug: Recording start even when device not connected

dheerajkotwani, 2020-01-26:
**Actual Behaviour**
Recording can be started even if the device is not connected to mobile or mobile does not have inbuilt feature support

**Expected Behaviour**
If any device is not connected then for that device recording should not start

**Steps to reproduce it**
Open any device which is not connected or whose sensor are not present in mobile(Gas Sensor)
Click on Record button
See the error

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

mariobehling, 2020-02-03:
Expected behavior would be a pop up "Please connect a device or choose an phone sensor (if available)" and two buttons "Configure Active Sensor" and "Cancel".

mariobehling, 2020-02-03:
In sensors that are usually present on the phone the recording should work nevertheless.

SpandanBasu, 2021-01-03:
I am a beginner. I want to work on this issue please assign me

SpandanBasu, 2021-01-03:
When the sensor is not present and the user clicks on the recording button, A pop-up dialog comes..right..but on pressing on Configure active sensor where should the app take the user?

sambhavsaxena, 2021-01-05:
Please assign this issue to me

SpandanBasu, 2021-01-05:
This issue is already fixed i opened a pull request. Try a different issue. Thanks

shubham-200315, 2024-01-07:
Hey @mariobehling , Issue is fixed and PR is merged already so i think we can close this.
Thank you.

marcnause, 2024-08-05:
Fixed in https://github.com/fossasia/pslab-android/pull/2295. A notification (SnackBar) is shown that informs the user that no device is connected.

closed 2024-08-05 by marcnause