fossasia/pslab-app #2099: Oscilloscope: Range does not change, when option is chosen

mariobehling, 2020-02-03:
The Oscilloscope has the option to change the range, e.g. of the Voltage display. This is currently not working. There is no visual change in the app if the user changes the range.

991rajat, 2020-02-03:
Works fine on my device.
[image]

Low012, 2020-02-04:
It works if the channel is selected (check mark is set). It does not work if the channel is not selected.

If the expected behaviour is that the range can only be changed if the channel is selected I would expect the range selection to be disabled if the channel is not selected. This would provide a visual hint to the user.

Also the range does not change if:
1. the channel is not selected
1. a different range is selected
1. the channel is selected

991rajat, 2020-02-05:
Still in my device.
If the channel is not selected -> changing range changes the graph.
If the channel is selected -> changing range changes the graph.
I'm not able to get it what bug is.

Low012, 2020-02-06:
It may depend on the Android API level. 

It works as pictured by @991rajat on my device with Android 9, but if I use my test device with Android 7 I get the behaviour as described by @mariobehling.

991rajat, 2020-02-09:
@Low012, Yaa not able to reproduce.

kartikeysaran, 2021-03-11:
Can i take up this issue

AnishaSingh0118, 2021-05-04:
may i work on this issue?

kartikeysaran, 2021-05-04:
Sure @AnishaSingh0118 go ahead !

AnishaSingh0118, 2021-05-04:
> Sure @AnishaSingh0118 go ahead !

Thanks 😊

marcnause, 2023-02-07:
I had a quick look at the issue again. I think it is only reproducable with the hardware. If no signal is displayed in the oscillosocope screen, changing the range works on my device. Once a signal is displayed, the range always resets to +/- 16V

marcnause, 2024-09-03:
I tested again with the latest development version of the app (https://github.com/fossasia/pslab-android/commit/5d6b449a6ca2d5b33ac1996f01f294a753cefaea) and a v6 board. The problem still exists.

PR #2552 (2024-11-04): refactor range select in oscilloscope
Fixes #2099

## Changes 
- refactoring of scale handling: moved scale values to extra class
- only set scale when required, not on every new data frame

**Checklist**: 
- [ ] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and fixed indentation in every file included in this pull request.
- [x] **No extra spa [... cut]
closed 2024-11-04 by marcnause