fossasia/pslab-app #2100: Wave Generator: triangle wave is not triangular, but cut off

mariobehling, 2020-02-03:
Wave Generator: This issue was reported by several users. The Triangle wave is not as it is supposed to be. The triangle wave is not triangular, but cut off.

Low012, 2020-02-04:
I think this is the same issue as https://github.com/fossasia/pslab-hardware/issues/111 which I opened in the hardware repo since I assumed it to be a hardware issue.

marcnause, 2023-02-06:
After plotting the values which are calculated in the `loadEquation(String channel, String function)` method of `io.pslab.communication.ScienceLab`, I am pretty confident that this is software issue.

The method includes a comment that the creation of different waveforms should be handled differently anyway. I think that using the "strategy" design pattern here could improve the code and also make the waveform creation testable in a unit test.

marcnause, 2023-02-06:
I fixed the wave, but I have not done any refactoring in `io.pslab.communication.ScienceLab`. This should be done in a second step.

PR #2364 (2023-10-30): Fix calculation of triangle wave
Fixes #2100

## Changes 
- fixed formula for calculation of triangle wave
- reformatted code
- optimized imports

## Screenshots / Recordings  
I did not change the UI, but this is what the wave looks like in the oscilloscope now.

[image]

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and  [... cut]
closed 2023-10-30 by bessman