fossasia/pslab-app #2108: Use Interface Callbacks to reduce Coupling between Activity and Fragments and reduce the lines of code

reckoner1429, 2020-02-19:
**Actual Behaviour**
`PsLabSensor` directly calls methods from fragments such as `LuxMeterDataFragment,` `SoundMeterFragment` etc, which increases coupling and also increases lines of codes in the `onOptionItemSelected()` method of `PsLabSensor`. In addition it increases chances of introducing bugs when new sensors are implemented. 

**Expected Behaviour**
**Interface callbacks** can be used to reduce the coupling between `PsLabSensor` and other data fragments. We can define a interface with common set of methods such as play(), stop(), pause(), resume() etc and have the fragments implement this interface.

**Steps to reproduce it**
`N/A`

**LogCat for the issue**

`N/A`

**Screenshots of the issue**
The lines of codes below are redundant.
[image]

**Would you like to work on the issue?**
Yes

VIKAS9899, 2020-05-14:
I want to work on this issue, Please assign it to @VIKAS9899

adityastic, 2020-05-14:
@VIKAS9899 go ahead

PR #2110 (2020-05-16): Fixes: PSLabSensor Activity decoupled from fragments
Fixes #2108

**Changes**: PSLabSensorActivity will interact with OperationCallback for operation not any specific fragments by doing this we can eliminate coupling & reduce boilerplate code.
closed 2020-05-16 by cweitat