fossasia/pslab-app #2126: Faq section expandable list view not working

CodeWithSouma, 2020-09-02:
## Faq section expandable list view not working when click on a question then the app was crash

### Expected Behaviour
It should show answer of that question 

### Screenshots of the issue

<pre>[image] [image] [image] [image]</pre>

### Gif 
<pre>
                                        [image] </pre>

ritmojs, 2020-09-13:
I want to work on this issue

CodeWithSouma, 2020-09-13:
> I want to work on this issue

I already **fixed** it. and send **PR** #2127

nileshteji, 2020-10-05:
@CodeWithSouma  the pr you mentioned is not mergest . Are you still working on this issue?Thank you

CodeWithSouma, 2020-10-05:
> Are you still working on this issue?

Yes i am woking on this issue

PR #2138 (2020-10-12): fix: FAQFragment crash #2126
@CloudyPadmal 
# Fix #2126 Faq expandable list view problem
Sir please check out my PR and merge it.
closed 2020-10-12 by CloudyPadmal