fossasia/pslab-app #2133: Update pin names

bessman, 2020-10-05:
Change all instances of

ID1/2/3/4 -> LA1/2/3/4
SEN -> RES
AN8 -> VOL
CNTR -> FRQ
W1/2 -> SI1/2

CodeWithSouma, 2020-10-06:
> Change all instances of
> 
> ID1/2/3/4 -> LA1/2/3/4
> SEN -> RES
> AN8 -> VOL
> CNTR -> FRQ
> W1/2 -> SI1/2
I don't  under stand where to change this instances....do you want to change these in string file ???

bessman, 2020-10-06:
Everywhere throughout the code. The old pin names were used in a previous hardware revision. For example, the pins called ID1, ID2, ID3, and ID4 in the code should be changed to LA1, LA2, LA3, and LA4 to reflect the current hardware.

This command should do the trick:
```
find app/src/main/ \( -type d -name .git -prune \) -o -type f -print0 | xargs -0 sed -i 's/ID1/LA1/g'
```
Repeat for all above-mentioned strings.

CodeWithSouma, 2020-10-06:
> Everywhere throughout the code. The old pin names were used in a previous hardware revision. For example, the pins called ID1, ID2, ID3, and ID4 in the code should be changed to LA1, LA2, LA3, and [...]

sir can you tell me how to use this command to find reference .....i don't  use this command before please help me i want to work on this issue

PR #2135 (2020-10-12): Update pin names
# Update pin names, Fix #2133 
Changed all instances of

ID1/2/3/4 -> LA1/2/3/4
SEN -> RES
AN8 -> VOL
CNTR -> FRQ
W1/2 -> SI1/2
commit 85fbd860 (2020-10-12): refactor: update pin names (#2135)

* refactor ID1 to LA1

* Refactor ID2 to LA2

* Refactor ID3 to LA3

* Refactor ID4 to LA4

* Refactor SEN to RES

* Refactor AN8 to VOL

* Refactor CNTR to FRQ

* Refactor W1 to SI1

* Refactor W2 to SI2

* Refactor code and fix SEN RES error

* refactor code

* refactored code

* Revert "Refactor W2 to SI2"

This reverts commit 75d27cac

* Revert "Refactor W1 to SI1"

This reverts commit 6f44370f

* Update pin names,fix #2133

* Rename W1 to SI1 and W2 to SI2
closed 2020-10-12 by CloudyPadmal