fossasia/pslab-app #2145: App crashing on Android 8.0

AtigPurohit, 2020-12-07:
[image]
**Actual Behaviour**

**Expected Behaviour**

**Steps to reproduce it**

**LogCat for the issue**

**Screenshots of the issue**

**Would you like to work on the issue?**

CodeWithSouma, 2020-12-14:
can you tell me when it crash

AtigPurohit, 2020-12-14:
@CodeWithSouma  the app is not opening itself, I gave the screen shot

CodeWithSouma, 2020-12-14:
> @CodeWithSouma the app is not opening itself, I gave the screen shot

I don't find any issue it runs smoothly on my device. I think you are using an emulator. Try to run on a physical device.

closed 2020-12-15 by AtigPurohit