fossasia/pslab-app #2172: Showing "Delete All Data" item menu even after all logs are deleted.

KushMehta1511, 2021-01-20:
**Actual Behaviour**

When we have deleted all the logs individually then also the menu item is showing "Delete All Data" option.

**Expected Behaviour**

Upon deletion of all the logs, the "Delete All Data" menu option should not be visible.

**Steps to reproduce it**

Create a log. Go to show logged data tab and delete the log using the delete icon. Open the items menu button. "Delete all Data" option still appears. 

**Screenshots of the issue**

https://user-images.githubusercontent.com/65716045/105130969-05359700-5b0e-11eb-9bc6-5ecc358c5e00.mp4

**Would you like to work on the issue?**

I tried to work upon the issue but was not successful in it. It would be of great help if the contributors could guide me through.

asaikarthikeya, 2021-01-20:
@KushMehta1511  Sure bro. What was your progress till now on this issue?

KushMehta1511, 2021-01-20:
@asaikarthikeya Thanks for the response. There are a few things which I could make out by analysing the code.
The activity of creating and maintaining the logs is being done in DataLoggerActivity.java and each data log is being stored in RealmReasults<SensorDataBlock> categoryData variable.
[image]
While filling the data to the recycler view it checks for the sie of the categoryData. But upon deletion of a data log, there is no place where the count is decremented. and hence it never the else part in the code snippet below.
[image]
The delete icon and its icons are present in SensorLoggerListAdapter.java class. In the handleDeleteItem method of SensorLoggerListAdapter the size of categoryData should be decremented so that the original count decreases.
[image]
I am not able to access the categoryData variable through the SensorLoggerListAdapter class. Also even if the count is decremented or the size of LocalDataLog.with().getAllSensorBlocks() becomes zero it won't reflect on the interface as the itemsMenu is being created only during onCreate() of the DataLoggerActivity. We instead would have to check the size of categoryData at each deletion and update the itemsMenu "Delete All Data" visibility instantaneously.
I am not able to perform this task of checking each time at deletion.
I tried to add a condition to check for LocalDataLog.with().getAllSensorBlocks().size() also in the necessary methods but apparently they are being created only once and hence the results are not getting affected.

Thanks for the help and please point out any flaws in the analysis.

KushMehta1511, 2021-01-23:
@CodeWithSouma Could you please guide me through this issue?

raghavpartani, 2021-03-30:
I want to work on this issue.

PR #2212 (2021-06-04): Fix:2172 Added change listener for Category Objects Size To update Delete all Menu Item Visibility
Fixes #2172 

**Changes**: Added a OnChangeListener that updates the visibility of the Delete All item in the menu

**Gif**: 
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR  [... cut]
closed 2021-06-04 by CloudyPadmal