fossasia/pslab-app #2173: Oscilloscope: After selecting PSLab MIC, In-Built-MIC does not work when PSLab device is not connected

arpoch, 2021-01-20:
**Actual Behaviour**
On selecting PSLab MIC option before the In-Built MIC option, no popup stating if a PSLab device is not connected, when selecting In-Build MIC no activity can be seen on the monitor.

**Expected Behaviour**
The PSLab MIC option should be unable when then PSLab device is not connected.

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

arpoch, 2021-01-22:
If have applied the changes, and the code is working fine. So should I know make a PR, can any please guide me through the procedure as it's my first PR.

[image]

asaikarthikeya, 2021-01-22:
Good , create a branch and push the changes to that branch. latter make  pull request to development branch .

asaikarthikeya, 2021-01-22:
Good. Create a new branch and push the changes to that branch. Later make pull request to development branch .

AsCress, 2024-05-26:
Superseeded by #2421 and resolved by #2422.

closed 2024-05-26 by AsCress