fossasia/pslab-app #2176: Keyboard hiding the snackbar output in USB Connection.

asaikarthikeya, 2021-01-21:
**Actual Behaviour**

When the user enters the IP Address of ESP module the error message is hidden behind the keyboard .User is able to see the snackbar output only if we press back button to remove keyboard 

**Expected Behaviour**

The error message should be visible to the user.

**Steps to reproduce it**

The edittext that is used to enter the IP address is replaced by materialdesign TextInputLayout ,so that we can view the error messages near the edittext instead of snackbar from bottom.

**Screenshots of the issue**

https://user-images.githubusercontent.com/70392921/105354591-c50e0b80-5c16-11eb-8374-7caa141a030c.mp4

KushMehta1511, 2021-01-25:
@asaikarthikeya I am not facing this issue on my device. Did you check using the emulator as well?

asaikarthikeya, 2021-01-26:
@KushMehta1511 Checked in 2 real device , Found this issue in both.

Prashant830, 2021-02-05:
Can I solve this problem ??

asaikarthikeya, 2021-02-05:
@Prashant830 PR is yet to be reviewed.

asaikarthikeya, 2021-02-05:
> Can I solve this problem ??

There is a PR already attach. Kindly wait till it gets reviewed.

Prashant830, 2021-02-06:
Ok

PR #2330 (2022-03-19): Fixed the snackbar issue
Fixes #2176 

## Changes 
After these changes, the keyboard collapses as soon as the "Connect" button is pressed to let the user see the snackbar.

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and fixed indentation in every file included in this pull request.
- [x] **No extra space**: My code does [... cut]
closed 2022-03-19 by CloudyPadmal