fossasia/pslab-app #2179: Repititive Appearance of toast message in Dust Sensor

KushMehta1511, 2021-01-23:
**Actual Behaviour**

In the dust sensor, the toast message showing "Not Connected" keeps on flashing and never disappears.

**Expected Behaviour**

The toast message should disappear after its stipulated duration.

**Steps to reproduce it**

Without connecting the device open the dust sensor. You will see the toast message repetitively flashing.

**Screenrecording of the issue**

https://user-images.githubusercontent.com/65716045/105568590-ab78db00-5d60-11eb-9939-ad0a6b1f30c3.mp4

CodeWithSouma, 2021-01-23:
@KushMehta1511 

**I don't find this type of issue on my device** 

[image]

KushMehta1511, 2021-01-24:
@CodeWithSouma ooh that's quite strange.  I have Redmi Note 5 pro with android version 8.1.0. Is it due to this that the behaviour is varying?

asaikarthikeya, 2021-01-24:
@KushMehta1511 Work fine with me too. I am not able to reproduce this issue .

KushMehta1511, 2021-01-24:
Ok, then I suppose there is some error from my side. Thanks, @CodeWithSouma @asaikarthikeya for the reviews. I will close this issue in that case.

closed 2021-01-24 by KushMehta1511