fossasia/pslab-app #2181: "Generate Config Files" layout is not uniform.

asaikarthikeya, 2021-01-23:
**Actual Behaviour**

The spacing between the instrument and time interval is varing for different instrument.

**Expected Behaviour**

All the instruments should have a similar layout style.

**Screenshots of the issue**

[image] [image] [image]

**Device**

Asus max pro m1(5.9 inch)

AnishaSingh0118, 2021-05-04:
hey, I am Anisha, I would like to work on this issue if it is still open.

kartikaysharma01, 2021-05-23:
> hey, I am Anisha, I would like to work on this issue if it is still open.

Hi anisha. This issue has a few open PRs which will potentially close it. You can look into some other issue. Thanks.

kartikaysharma01, 2021-05-23:
@CloudyPadmal can we close and move this issue to #2208 to reduce confusion as a whole layout change has been decided

AnishaSingh0118, 2021-05-30:
> > hey, I am Anisha, I would like to work on this issue if it is still open.
> 
> Hi anisha. This issue has a few open PRs which will potentially close it. You can look into some other issue. Thanks.

Oh ok :)

PR #2210 (2021-06-04): Fixes #2208 Card Layout Implemented for Generate Config Layout
Fixes #2208 
Fixes #2181 

**Changes**:
1. Added Card Views for Instrument and Time Interval Sections on Generate Config Screen
2. Regularized the layout

**Screenshot/s for the changes**:
[image]
[image]

**Checklist**: [Please tick following check boxes with `[x]` if the respective task is completed]
- [x] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code [... cut]
closed 2021-06-04 by CloudyPadmal