fossasia/pslab-app #2186: App crashes when searching for bluetooth devices

KushMehta1511, 2021-01-26:
**Actual Behaviour**

While scanning for Bluetooth devices the app crashes.

**Expected Behaviour**

There should not be any crash.

**Steps to reproduce it**

Click on the USB icon on the main page. Select the Bluetooth option and start the scan. After a while, no devices would be found and the app crashes.

**Screenshots of the issue**

https://user-images.githubusercontent.com/65716045/105789837-25030a00-5fa9-11eb-8a8f-106d6921a58e.mp4

CodeWithSouma, 2021-01-26:
I don't find any type of crash issue on click the Bluetooth button ---> scan button.

asaikarthikeya, 2021-01-26:
Me too, Didn't find this issue.Works as expected .

kartikeysaran, 2021-01-27:
Hi @KushMehta1511 , Works as expected can you please share your log while the crash occurs

KushMehta1511, 2021-01-27:
OK Thank You everyone for confirming. I will close the issue in that case.

closed 2021-01-27 by KushMehta1511

Low012, 2021-09-20:
I can reproduce the issue. My devices is a Fairphone 2 with LineageOS 18.1 (Android 11). Bluetooth is switched on, but no device is connected to the phone.

`2021-09-20 21:07:36.098 1500-1500/io.pslab E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.pslab, PID: 1500
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.lang.Object.toString()' on a null object reference
        at android.widget.ArrayAdapter.createViewFromResource(ArrayAdapter.java:454)
        at android.widget.ArrayAdapter.getView(ArrayAdapter.java:416)
        at android.widget.AbsListView.obtainView(AbsListView.java:2387)
        at android.widget.ListView.measureHeightOfChildren(ListView.java:1417)
        at android.widget.ListView.onMeasure(ListView.java:1323)
        at android.view.View.measure(View.java:25466)
        at android.widget.RelativeLayout.measureChildHorizontal(RelativeLayout.java:735)
        at android.widget.RelativeLayout.onMeasure(RelativeLayout.java:481)
        at android.view.View.measure(View.java:25466)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6957)
        at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
        at android.view.View.measure(View.java:25466)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6957)
        at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
        at android.view.View.measure(View.java:25466)
        at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6957)
        at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
        at com.android.internal.policy.DecorView.onMeasure(DecorView.java:747)
        at android.view.View.measure(View.java:25466)
        at android.view.ViewRootImpl.performMeasure(ViewRootImpl.java:3402)
        at android.view.ViewRootImpl.measureHierarchy(ViewRootImpl.java:2246)
        at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2504)
        at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1948)
        at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:8177)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:972)
        at android.view.Choreographer.doCallbacks(Choreographer.java:796)
        at android.view.Choreographer.doFrame(Choreographer.java:731)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:957)
        at android.os.Handler.handleCallback(Handler.java:938)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:223)
        at android.app.ActivityThread.main(ActivityThread.java:7664)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)`

I suggest to reopen this ticket.

Let me know if I can do anything else to help pinning down the issue.

Low012, 2021-09-20:
I found the problem: Android calls `toString()` on `null` when a Bluetooth device has no name. I will fix the issue and open a pull request.

PR #2223 (2021-09-21): Fix: 2186 Fix crash when searching for bluetooth devices
Fixes #2186

**Changes**: Provide fallback if a discovered Bluetooth device has no name.

**Checklist**:
- [ ] I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard-coding them
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines or extra spaces
- [ ] I have requested reviews from other members

**APK for  [... cut]