fossasia/pslab-app #2188: Time interval units are not shown to the user.

asaikarthikeya, 2021-02-01:
**Actual Behaviour**

Time interval units are not known to the user when generating a config file.

**Expected Behaviour**

User should be able to know what time interval is selected.

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes

rohittrimalle, 2021-02-02:
Can I work on this issue ?

asaikarthikeya, 2021-02-02:
> Can I work on this issue ?

Sorry.I started working on this. Please find another issue. Thankyou

nsharmahack3r, 2021-02-10:
https://github.com/fossasia/pslab-android/pull/2192

Issue solved

kartikeysaran, 2021-05-12:
This needs a complete layout change i am working on this

kartikaysharma01, 2021-05-23:
@CloudyPadmal  can we close and move this issue to #2208 to reduce confusion as a whole layout change has been decided

marcnause, 2024-06-23:
Checked with the most recent version and units are still not shown in vertical screen orientation.

[image]

AsCress, 2024-06-25:
@marcnause Ahh yes. I see. The issue is the dimension and the arrangement of the elements in there. I'll fix this in a PR soon.

Dishebh, 2024-09-09:
hey @AsCress , if you haven't yet started work on this yet, do you mind if I could take this issue?

AsCress, 2024-09-10:
@Dishebh Yeah sure ! Please go ahead !

PR #2672 (2025-04-28): fix: spinner hint display
Fixes #2188 
## Changes 
Changed layout from LinearLayout to ConstraintLayout.
## Screenshots / Recordings  
**Before**
[image]
**After**
[image]

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and fixed indentation in every file included in this pull request.
- [x] **No extra space**: My code does [... cut]
closed 2025-04-28 by marcnause