fossasia/pslab-app #2190: At navigation drawer, The menu option of Report a Bug is missing ..

Prashant830, 2021-02-04:
**Actual Behaviour**
At navigation drawer, The menu option of Report a Bug is missing ..

**Expected Behaviour**
After creating the report a bug option, navigation drawer, if any user sees any issue or bug in app, then that user will click on this option and then he will come to the issue option of github repository of app so that he can create his new issue .

**Steps to reproduce it**
Creating a menu option for report a bug at navigation drawer.

**Screenshot**
[image]

**Would you like to work on the issue?**
yes

MonuGit9, 2024-01-13:
Claim

marcnause, 2024-09-03:
I am closing this issue since the PR for this issue was closed without merge with the reasoning that adding more options would overcrowd the navigation drawer,

closed 2024-09-03 by marcnause