fossasia/pslab-app #2206: getDrawable() is deprecated and needs to be replaced

Abhinavtdk, 2021-03-20:
**Actual Behaviour**

getDrawable( int ) is deprecated and is used in many places in the project

**Expected Behaviour**

getDrawable() needs to be replaced, we can use ResourcesCompat.getDrawable() 

**Would you like to work on the issue?**

Yes

Glock123, 2021-03-26:
Is this issue closed?

Abhinavtdk, 2021-03-26:
> Is this issue closed?

I made a pr #2207 , it has to be approved

Aditya8840, 2021-08-02:
is this open ?

kartikaysharma01, 2021-08-02:
Hi @Aditya8840, This issue is not a priority right now. Please have a look at some other issues. You can refer to the prior discussion [here](https://github.com/fossasia/pslab-android/pull/2207#issuecomment-811760004). Thanks.

sakshampruthi, 2021-10-15:
Great, I'd like to work on this issue

Prakhar-Agarwal-byte, 2021-11-22:
I would like to work on this issue

Prakhar-Agarwal-byte, 2021-11-22:
I am working on this issue

closed 2022-02-01 by CloudyPadmal