fossasia/pslab-app #2209: Offer choice of Google Maps or OpenStreetMap

kartikaysharma01, 2021-04-01:
**Actual Behaviour**
For Playstore users, we can provide an option to use Google Maps or OpenStreetMap in App settings 

**Would you like to work on the issue?**
Yes

kartikaysharma01, 2021-04-01:
@CloudyPadmal Should i go ahead with the implementation ?

CloudyPadmal, 2021-04-01:
Yes please go ahead.

kartikaysharma01, 2021-04-01:
@CloudyPadmal What should be the default map for playstore users?  
Should we keep OpenStreetMap as default and give a small one-time-warning before switching to Google Maps that it is a proprietary API.

kartikeysaran, 2021-05-11:
I am working on this

kartikaysharma01, 2021-05-23:
I am occupied in another project as of now. If somebody wants to pick up this issue you are most welcome.

sempsmahad, 2021-12-14:
Hello guys, I am new to the technology, on which screen is the current version of the map?, I can not find it.
Adding the options in settings should be easily doable but the effect on the interface

CloudyPadmal, 2022-01-31:
Not using Google Maps anymore. Moved to Openstreetmaps.

closed 2022-01-31 by CloudyPadmal