fossasia/pslab-app #2211: Find a crash

LvZW1895, 2021-04-07:
APK Version : 2.0.20
AndroidOS Version: 7.1.2
Description information：
This crash happened by accident when I was testing in different devices. I tried to reproduce them and found that they can't be reproduced, but they are real. I hope they can help you to debug.

crash:
 CRASH: io.pslab (pid 15648) (dump time: 2021-03-22 15:05:08
 Version: 2.0.20Version:
 Long Msg: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.view.View.setVisibility(int)' on a null object reference
 	at io.pslab.fragment.HomeFragment$1$1.onPageFinished(HomeFragment.java:125)
 	at u44.b(chromium-Monochrome.aab-stable-438909023:2)
 	at Ct.handleMessage(chromium-Monochrome.aab-stable-438909023:65)
 	at android.os.Handler.dispatchMessage(Handler.java:102)
 	at android.os.Looper.loop(Looper.java:154)
 	at android.app.ActivityThread.main(ActivityThread.java:6121)
 	at java.lang.reflect.Method.invoke(Native Method)
 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
 crashend

Abhinavtdk, 2021-04-07:
I'll try working on this

faisalkhandholpur, 2021-05-03:
I can handle this issue

kartikeysaran, 2021-05-12:
I am not able to reproduce this as i can see it can be either due to the visibility of the webview or the visibility of progress bar but both are initialized if you are able to reproduce this please ping me. Thanks !

Low012, 2021-09-21:
I managed to trigger a crash in the same class:

>     java.lang.NullPointerException: Attempt to invoke virtual method 'void android.webkit.WebView.loadUrl(java.lang.String)' on a null object reference
>        at io.pslab.fragment.HomeFragment$1.onClick(HomeFragment.java:111)
>        at android.view.View.performClick(View.java:7259)
>        at android.view.View.performClickInternal(View.java:7236)
>        at android.view.View.access$3600(View.java:801)
>        at android.view.View$PerformClick.run(View.java:27900)
>        at android.os.Handler.handleCallback(Handler.java:883)
>        at android.os.Handler.dispatchMessage(Handler.java:100)
>        at android.os.Looper.loop(Looper.java:214)
>        at android.app.ActivityThread.main(ActivityThread.java:7559)
>        at java.lang.reflect.Method.invoke(Native Method)
>        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
>        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:935)

The crash does not happen in exactly the same location as the crash reported by @LvZW1895, but it happens just a few lines away. `webview`should also be not `null` here, but under certain conditions it is.

I was able to trigger the crash on my tablet (Lenovo Yoga Smart Tab, Android 10):
1. open app
2. choose "connect device' in the menu on the left
3. hover one finger over 'connect device'
4. hover another finger over 'What is PSLab Device'
5. tap both fingers without changing the positions of the fingers on the screen rapidly and repeatedly

I propose adding `null`-checks in the methods `onClick`, `onPageStarted` and `onPageFinished`.

If nobody else wants to work on this I can fix the issue and create a pull request.

PR #2224 (2021-09-22): Fixes #2211 fix rare crash
Fixes #2211

**Changes**: Added several null-checks to prevent NullPointerExceptions when referned views have already been removed in a later stage of the Fragment lifecycle.

**Checklist**: 
- [x] No modifications done at the end of resource files `strings.xml`, `dimens.xml` or `colors.xml`
- [x] I have reformatted code in every file included in this PR [<kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>L</kbd>]
- [x] My code does not contain any extra lines or extra spaces
- [ ] I have requested reviews from other members

**APK for testing**:
[fix_rare_crash.zip](https://github [... cut]
commit 8c7b02ad (2021-09-22): Fixes #2211 fix rare crash (#2224)
closed 2021-09-22 by CloudyPadmal