fossasia/pslab-app #2217: Bug in connect device activity.

Prashant830, 2021-05-31:
**Actual Behaviour**

Bug in connect device activity.

**Expected Behaviour**

When i click <b>`What is psLab device ? textView`</b> and at the same time i press backspace button then the fragment of pslab device overloaded at the connect device activity due to this the ui of the connect device activity is mismatching form (It Ui looks not batter ).

**Steps to reproduce it**
When i click <b>`What is psLab device ? textView`</b> and at the time i press backspace button then it not showing any further activity ,simply showing connect device ui..

**Screen Recording of the issue**

[Click here for access Screen Recording Video ](https://drive.google.com/file/d/1uONkhBF46c4Yqh3mu5SC7bpmIzTwX6TJ/view?usp=drivesdk) 

**Would you like to work on the issue?**
Yes I would like to solve this issue..

Prashant830, 2021-05-31:
@mariobehling , @CodeWithSouma please review it..

CloudyPadmal, 2021-05-31:
Thanks for reporting the issue. If you can fix it, please go ahead.

Prashant830, 2021-05-31:
@CloudyPadmal ok..

Prashant830, 2021-06-02:
@CloudyPadmal , @CodeWithSouma   [ please review  solution  in sample videos](https://drive.google.com/drive/folders/1wE6W-_STez0wA_st5AGYEAttWbUMf2iI?usp=sharing)

CloudyPadmal, 2021-06-02:
Looks good. Can you file a pull request with the modifications?

PR #2332 (2022-04-06): Fixed the bug in connect device activity
Fixes #2217 

## Changes 
- 

## Screenshots / Recordings  

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and fixed indentation in every file included in this pull request.
- [x] **No extra space**: My code does not contain any extra lines or extra spaces than the ones that are nec [... cut]
closed 2022-04-06 by CloudyPadmal