fossasia/pslab-app #2219: Threat of cash with NoClassDefFoundError on some devices

Linlinllll, 2021-07-06:
Dear pslab developers,

I have found out that when I touch any point on the UI screen shown bellow, the app will crash with `NoClassDefFoundError`. After I check the log file, I found the method is `android/text/Html.fromHtml(Ljava/lang/String;I)Landroid/text/Spanned;`. This method is introduced at API level 24, but the `minSDKVeriosn` is set to 23. When I install the app into devices with API level 23, and then touch any point on the screen shown bellow, the app will crash with `NoClassDefFoundError`.

Thank you very much if you can fix this. Hope this information is helpful. :)

[image]

Low012, 2022-03-24:
I think this issue should be fixed in version 2.1.0

CloudyPadmal, 2022-03-25:
Thanks @Low012 for following this up. Let me check if it still exists and update this issue.

CloudyPadmal, 2022-03-25:
This is fixed

closed 2022-03-25 by CloudyPadmal