fossasia/pslab-app #2242: Upgrade layout files with android-x compatible tags

CloudyPadmal, 2021-10-06:
**Actual Behaviour**

Most of the `.xml` files have outdated syntax

**Expected Behaviour**

Need to replace them with android-x tags

**Steps to reproduce it**

Open app and check error log

**LogCat for the issue**

```
Process: io.pslab, PID: 7029
    android.view.InflateException: Binary XML file line #9 in io.pslab:layout/application_list_item: Binary XML file line #9 in io.pslab:layout/application_list_item: Error inflating class android.support.v7.widget.CardView
    Caused by: android.view.InflateException: Binary XML file line #9 in io.pslab:layout/application_list_item: Error inflating class android.support.v7.widget.CardView
    Caused by: java.lang.ClassNotFoundException: android.support.v7.widget.CardView
...
```

**Screenshots of the issue**

N/A

**Would you like to work on the issue?**

Yes; but also open to anyone if like.

SiddyDevelops, 2021-10-08:
Is this issue still open for contributors? If yes, can you please assign it to me ?  Thank you.

MohitGupta121, 2021-10-09:
@CloudyPadmal  hello I also started working on that issue

MohitGupta121, 2021-10-09:
@CloudyPadmal can you please review my PR as I'll change some of the files to androidx according to issue.

CloudyPadmal, 2021-10-09:
Hello @deepanshu112 , great to hear that! You're welcome to send us a PR.

closed 2022-02-01 by CloudyPadmal