fossasia/pslab-app #2245: Create auto builds of APKs using Github Actions

mariobehling, 2021-10-10:
Until a few months ago Travis supported us and we used it for continuous integration and creating auto-builds of APKs in the apk branch. Please update the project to use Github Actions instead of Travis CI to create these auto builds in https://github.com/fossasia/pslab-android/tree/apk.

MohitGupta121, 2021-10-10:
Okay I'm ready to work on that issue @mariobehling can you please assign that issue.

MohitGupta121, 2021-10-10:
@mariobehling  I have make initial PR for this issue so that I can step by step work on that issue.
I need guide to work on this issue can you please Review my this PR and tell me what more things required in this PR so that I can add that also.

CloudyPadmal, 2021-10-13:
It's not solved to say. But the APKs are available as artefacts. We might still need to implement this feature

CloudyPadmal, 2022-01-31:
Fixed with #2314

closed 2022-01-31 by CloudyPadmal