fossasia/pslab-app #2280: App crash on opening ROBOTIC ARM

suman-somu, 2021-11-06:
**Actual Behaviour**

On opening Robotic arm, the app crashes

**Expected Behaviour**

App should not crash

**Steps to reproduce it**

try to open ROBOTIC ARM

**LogCat for the issue**

FATAL EXCEPTION: main
    Process: io.pslab, PID: 7970
    java.lang.RuntimeException: Unable to start activity ComponentInfo{io.pslab/io.pslab.activity.RoboticArmActivity}: android.view.InflateException: Binary XML file line #9 in io.pslab:layout/activity_robotic_arm: Binary XML file line #9 in io.pslab:layout/activity_robotic_arm: Error inflating class android.support.design.widget.CoordinatorLayout
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3735)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3903)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2328)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:255)
        at android.app.ActivityThread.main(ActivityThread.java:8215)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:632)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049)
     Caused by: android.view.InflateException: Binary XML file line #9 in io.pslab:layout/activity_robotic_arm: Binary XML file line #9 in io.pslab:layout/activity_robotic_arm: Error inflating class android.support.design.widget.CoordinatorLayout
     Caused by: android.view.InflateException: Binary XML file line #9 in io.pslab:layout/activity_robotic_arm: Error inflating class android.support.design.widget.CoordinatorLayout
     Caused by: java.lang.ClassNotFoundException: android.support.design.widget.CoordinatorLayout
        at java.lang.Class.classForName(Native Method)
        at java.lang.Class.forName(Class.java:454)
        at android.view.LayoutInflater.createView(LayoutInflater.java:813)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:1008)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:959)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:657)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:532)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:479)
        at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:699)
        at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:195)
        at io.pslab.activity.RoboticArmActivity.onCreate(RoboticArmActivity.java:109)
        at android.app.Activity.performCreate(Activity.java:8151)
        at android.app.Activity.performCreate(Activity.java:8135)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1309)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3704)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3903)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2328)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:255)
        at android.app.ActivityThread.main(ActivityThread.java:8215)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:632)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049)
    Caused by: java.lang.ClassNotFoundException: Didn't find class "android.support.design.widget.CoordinatorLayout" on path: DexPathList[[zip file "/data/app/~~welsSkYPmsxgPdw7AGJ7Vw==/io.pslab-gn6-_DIOqzhQURc7bTSLAQ==/base.apk"],nativeLibraryDirectories=[/data/app/~~welsSkYPmsxgPdw7AGJ7Vw==/io.pslab-gn6-_DIOqzhQURc7bTSLAQ==/lib/arm64, /data/app/~~welsSkYPmsxgPdw7AGJ7Vw==/io.pslab-gn6-_DIOqzhQURc7bTSLAQ==/base.apk!/lib/arm64-v8a, /system/lib64, /system/system_ext/lib64, /vendor/lib64, /odm/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:207)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        	... 26 more

**Screenshots of the issue**

https://user-images.githubusercontent.com/75962762/140606255-c119a9b5-07d8-4a81-b5ae-8b8ca4433354.mp4

**Would you like to work on the issue?**
No

suman-somu, 2021-11-07:
done this fix .
will be submitting a pr.

CloudyPadmal, 2021-11-07:
Fixed in #2281

closed 2021-11-07 by CloudyPadmal