fossasia/pslab-app #2285: Bug in Wave Generator

suman-somu, 2021-11-15:
**Actual Behaviour**
We have to click twice to produce sound 

**Expected Behaviour**
Should produce sound in one click

**Steps to reproduce it**
Open WAVE GENERATOR - try to click PRODUCE SOUND 

**LogCat for the issue**

```2021-11-15 08:19:09.266 11836-11836/io.pslab D/ViewRootImpl: enqueueInputEventMotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=233.5, y[0]=1497.5, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89772207, downTime=89772207, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:09.267 11836-11836/io.pslab D/ViewRootImpl[MainActivity]: processMotionEvent MotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=233.5, y[0]=1497.5, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89772207, downTime=89772207, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:09.268 11836-11836/io.pslab D/ViewRootImpl[MainActivity]: dispatchPointerEvent handled=true, event=MotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=233.5, y[0]=1497.5, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89772207, downTime=89772207, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:09.327 11836-11836/io.pslab D/ViewRootImpl[MainActivity]: processMotionEvent MotionEvent { action=ACTION_UP, actionButton=0, id[0]=0, x[0]=233.5, y[0]=1497.5, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89772267, downTime=89772207, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:09.329 11836-11836/io.pslab D/ViewRootImpl[MainActivity]: dispatchPointerEvent handled=true, event=MotionEvent { action=ACTION_UP, actionButton=0, id[0]=0, x[0]=233.5, y[0]=1497.5, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89772267, downTime=89772207, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:09.342 11836-11836/io.pslab E/Parcel: Reading a NULL string not supported here.
2021-11-15 08:19:09.367 11836-12115/io.pslab E/libEGL: Invalid file path for libcolorx-loader.so
2021-11-15 08:19:09.367 11836-12115/io.pslab D/libMEOW: applied 1 plugins for [io.pslab]:
2021-11-15 08:19:09.367 11836-12115/io.pslab D/libMEOW:   plugin 1: [libMEOW_gift.so]:
2021-11-15 08:19:09.524 11836-11836/io.pslab W/BottomSheetBehavior: BottomSheetBehavior now supports multiple callbacks. `setBottomSheetCallback()` removes all existing callbacks, including ones set internally by library authors, which may result in unintended behavior. This may change in the future. Please use `addBottomSheetCallback()` and `removeBottomSheetCallback()` instead to set your own callbacks.
2021-11-15 08:19:09.566 11836-11836/io.pslab V/PhoneWindow: DecorView setVisiblity: visibility = 4, Parent = null, this = DecorView@b611ad6[]
2021-11-15 08:19:09.568 11836-11836/io.pslab D/ColorViewRootUtil: initSwipState, isDisplayCompatApp false
2021-11-15 08:19:09.568 11836-11836/io.pslab D/ColorViewRootUtil: mScreenHeight 2400, mScreenWidth 1080
2021-11-15 08:19:09.572 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: hardware acceleration = true , fakeHwAccelerated = false, sRendererDisabled = false, forceHwAccelerated = false, sSystemRendererDisabled = false
2021-11-15 08:19:09.576 11836-11836/io.pslab E/libEGL: Invalid file path for libcolorx-loader.so
2021-11-15 08:19:09.576 11836-11836/io.pslab I/chatty: uid=10277(io.pslab) identical 4 lines
2021-11-15 08:19:09.576 11836-11836/io.pslab E/libEGL: Invalid file path for libcolorx-loader.so
2021-11-15 08:19:09.577 11836-11836/io.pslab I/InputTransport: Create ARC handle: 0x7304319ae0
2021-11-15 08:19:09.577 11836-11836/io.pslab V/PhoneWindow: DecorView setVisiblity: visibility = 0, Parent = android.view.ViewRootImpl@cdf492d, this = DecorView@b611ad6[WaveGeneratorActivity]
2021-11-15 08:19:09.636 11836-11836/io.pslab V/OplusZoomWindowDecorViewHelper: setLastReportedMergedConfiguration mZoomDisplayHeight: 2400 getDecorView.DecorView@b611ad6[WaveGeneratorActivity]
2021-11-15 08:19:10.034 11836-11836/io.pslab V/PhoneWindow: DecorView setVisiblity: visibility = 4, Parent = android.view.ViewRootImpl@d18ddd0, this = DecorView@51ad5cd[MainActivity]
2021-11-15 08:19:11.194 11836-11836/io.pslab D/ViewRootImpl: enqueueInputEventMotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=158.125, y[0]=1341.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89774133, downTime=89774133, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:11.194 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: processMotionEvent MotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=158.125, y[0]=1341.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89774133, downTime=89774133, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:11.196 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: dispatchPointerEvent handled=true, event=MotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=158.125, y[0]=1341.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89774133, downTime=89774133, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:11.309 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: processMotionEvent MotionEvent { action=ACTION_UP, actionButton=0, id[0]=0, x[0]=158.125, y[0]=1341.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89774247, downTime=89774133, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:11.310 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: dispatchPointerEvent handled=true, event=MotionEvent { action=ACTION_UP, actionButton=0, id[0]=0, x[0]=158.125, y[0]=1341.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89774247, downTime=89774133, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:12.771 11836-11836/io.pslab D/ViewRootImpl: enqueueInputEventMotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=145.625, y[0]=1366.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89775711, downTime=89775711, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:12.771 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: processMotionEvent MotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=145.625, y[0]=1366.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89775711, downTime=89775711, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:12.772 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: dispatchPointerEvent handled=true, event=MotionEvent { action=ACTION_DOWN, actionButton=0, id[0]=0, x[0]=145.625, y[0]=1366.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89775711, downTime=89775711, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:12.944 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: processMotionEvent MotionEvent { action=ACTION_UP, actionButton=0, id[0]=0, x[0]=145.625, y[0]=1366.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags=0x2, edgeFlags=0x0, pointerCount=1, historySize=0, eventTime=89775858, downTime=89775711, deviceId=4, source=0x1002, displayId=0 }
2021-11-15 08:19:12.946 11836-11836/io.pslab D/ViewRootImpl[WaveGeneratorActivity]: dispatchPointerEvent handled=true, event=MotionEvent { action=ACTION_UP, actionButton=0, id[0]=0, x[0]=145.625, y[0]=1366.625, toolType[0]=TOOL_TYPE_FINGER, buttonState=0, classification=NONE, metaState=0, flags [... cut ...]
2021-11-15 08:19:23.809 11836-11836/io.pslab V/PhoneWindow: DecorView setVisiblity: visibility = 4, Parent = android.view.ViewRootImpl@24286e2, this = DecorView@ff237d7[WaveGeneratorActivity]
2021-11-15 08:19:23.812 11836-11836/io.pslab D/View: [Warning] assignParent to null: this = DecorView@ff237d7[WaveGeneratorActivity]
2021-11-15 08:19:23.818 11836-11836/io.pslab I/InputTransport: Destroy ARC handle: 0x72a68b9500
```

**Screenshots of the issue**

https://user-images.githubusercontent.com/75962762/141715819-7d6ec9ee-ae1d-4598-9c12-d173b3a6b763.mp4

**Would you like to work on the issue?**
yes, but anyone else can too.

suman-somu, 2021-11-16:
This is not only for wavegenerator but for every option. We have to click twice to get started everytime.

**Steps to reproduce -** try to open any option and start the respective function in one click
<br>
**Screenshots-**

https://user-images.githubusercontent.com/75962762/141953132-acf81642-610f-4184-8d90-8e69cd7949e8.mp4

<br> 
@CloudyPadmal , I want some help in this. I don't know,  even is this a issue or not.

CloudyPadmal, 2021-11-16:
Hello @suman-somu , this is not the intended behaviour for the buttons and we can call this is a bug. I suspect there is a view overlapping the who UI that prevents the first click. 🤔 but I’m not 100% sure if that’s the case.

suman-somu, 2021-11-16:
Yep, I too think that might be the reason for this.

Low012, 2022-03-24:
The reason for this behaviour seems to be the OnClickListener which is added to tvShadow in line 245 of the WaveGeneratorActivity. Once lines 245 to 252 are removed, the button click behaviour is as desired (but the BottomSheet cannot be hidden by clicking the shadow layer anymore).

I will try to fix this, but not tonight. 😴

PR #2331 (2022-03-29): fix: extract bottom sheet handling and fix related problems
Fixes #2285

## Changes 
- Extracted bottom sheet handling to GuideActivity
- Let Activities extend GuideActivity where applicable
- Fixed problem with buttons which have to be clicked twice (visibility of shadow layer was not set to `GONE` when alpha value was 0)
- Reformatted several files

## Screenshots / Recordings  
https://user-images.githubusercontent.com/11596220/160490537-cc7c13a9-ab9f-462d-b0c8-254e35373386.mp4

**Checklist**: 
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `di [... cut]
closed 2022-03-29 by CloudyPadmal