fossasia/pslab-app #2302: Replace Google maps with open street maps

CloudyPadmal, 2022-01-17:
**Actual Behaviour**

There are two bud flavours, one using google maps and the other using open maps.

**Expected Behaviour**

We can steer away from Google maps altogether by replacing that with openstreetmap https://www.openstreetmap.org/#map=2/70.0/-10.5

**Steps to reproduce it**

> feature request

**LogCat for the issue**

> feature request

**Screenshots of the issue**

> feature request

**Would you like to work on the issue?**

Open for new contributors

hrishikeshnikam2000, 2022-01-18:
May I work on this issue? 

If yes this would be my first PR to this repo, are there any things I should keep in mind while executing this PR?

I had a look at pslab-android/app/src/main/java/io/pslab/others/GPSLogger.java file and I could see how Location will be logged, and I suppose I will start working on this file right?

CloudyPadmal, 2022-01-18:
Hello @hrishikeshnikam2000 , yes you can work on this issue and seems like you are on the right path

closed 2022-01-29 by CloudyPadmal