fossasia/pslab-app #2342: Multimeter sceen layout incomplete

Low012, 2022-04-13:
**Actual Behaviour**
The multimeter screen shows only part of the layout. The knob is not visible.

**Expected Behaviour**
The complete layout should be visible, including the knob.

**Steps to reproduce it**
Open the multimeter screen

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Yes

rehanwick, 2022-08-29:
## Please Assigned me this issue 
I Have seen the issue- 
- XMl  file error Layout 
- activity_ultimeter_main.xml file the layout error 

I can solve the issue please Assigned me the task

Low012, 2022-08-29:
Hey @rehanwick, thanks for having a look. I wanted to fix the issue, but my Android layout skills are a little bit rusty (to say the least) and I didn't manage to solve the problem. I don't have the access rights to assign you this task, but I am not working on it, so you won't be doing double work.

MonuGit9, 2024-01-13:
Claim

PR #2407 (2024-05-16): fix: updated multimeter screen layout to make it visible properly
Fixes #2342 and #2394 
This issue was arising due to a redundant View bottom_view being present in the layouts, something which has already been implemented in activity_multimeter_main.xml.
## Changes 
- app/src/main/res/layout-hdpi/activity_multimeter.xml,
app/src/main/res/layout-sw600dp/activity_multimeter.xml,
app/src/main/res/layout-xhdpi/activity_multimeter.xml,
app/src/main/res/layout/activity_multimeter.xml - Removed bottom_view and fixed some other constraints, hardcoding.
- app/src/main/res/values-hdpi/dimens.xml,
 app/ [... cut]
closed 2024-05-16 by CloudyPadmal