fossasia/pslab-app #2352: Compatibility issue when calling method

PSDroid2022, 2022-08-22:
We confirm a compatibility issue which might threaten the robustness of your app and give a detailed suggestion for you.

In ''io.pslab.activity.LuxMeterActivity", you invoke the framework API "<android.os.BaseBundle: long getLong(java.lang.String)>" in "getDataFromDataLogger" method as shown in following. But actually, this method is introduced after API level 21 (https://developer.android.google.cn/reference/kotlin/android/os/BaseBundle?hl=en#getLong(kotlin.String)).

```java
  @Override
    public void getDataFromDataLogger() {
        if (getIntent().getExtras() != null && getIntent().getExtras().getBoolean(KEY_LOG)) {
            //playingData = true;
            viewingData = true;
            recordedLuxData = LocalDataLog.with()
                    .getBlockOfLuxRecords(getIntent().getExtras().getLong(DATA_BLOCK));
            final LuxData data = recordedLuxData.get(0);
            if (data != null) {
                final String title = titleFormat.format(data.getTime());
                final ActionBar actionBar = getSupportActionBar();
                if (actionBar != null) {
                    actionBar.setTitle(title);
                }
            }
        }
    }
	}
``` 
So when the app try to invoke this API on devices before API level 21, your app will run with an unpredictable results. So we suggest you add an "if(SDK_INT >= 21)" to fix this potential issue. 

## Android device

 - Device: [e.g. Emulator]
 - OS: [e.g. API 19]
 - PSLab Version: [2.0.10]

install app on devices with API 19, open PSLab with crash. 

**Would you like to work on the issue?**
Yes

beradeep, 2023-07-23:
What if we use `getLongExtra()` (added in API lvl 1) intead of `getExtras().getLong()` (added in API lvl 21) for this case?

beradeep, 2023-10-16:
Hey @PSDroid2022, please check this out.

PR #2370 (2023-10-29): fix: Compatibility issue while calling getDataFromDataLogger() method
## Fixes #2352

## Changes:
- Added an SDK check to prevent crashes/unpredictable results when calling the method on devices below API 21

## Screenshot/Recordings:
_Not applicable._

## Checklist: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and fixed indentation in  [... cut]
closed 2023-10-29 by mariobehling