fossasia/pslab-app #2417: Update app on Fdroid

mariobehling, 2024-05-17:
Update app on Fdroid https://f-droid.org/en/packages/io.pslab/

AsCress, 2024-05-17:
Thank you for assigning me this issue. I just wanted to ask for a bit of clarification on this. By updating, do you mean releasing a newer version of this on FDroid. Would I be needing any sort of credentials for that ?

CloudyPadmal, 2024-05-17:
@AsCress I think most of the details, I have logged here in this blog post https://blog.fossasia.org/publish-an-open-source-app-on-fdroid/

CloudyPadmal, 2024-05-17:
and yea it's about making a new release. Please let me know if you encounter any issues. Thanks!

AsCress, 2024-05-18:
Thank you very much @CloudyPadmal for all that info and your help!! 
I am getting this error when I try to run fdroid lint, fdroid rewritemeta or fdroid build:
2024-05-18 14:22:12,002 CRITICAL: '0' is not a valid Archive Policy in org.fossasia.pslab. Regex pattern: ^[0-9]+ versions$
I have been trying to resolve this since quite some time but in vain.

AsCress, 2024-05-18:
@CloudyPadmal That issue is resolved. I guess it had something to do with my local installation of Ubuntu. However, when I now run fdroid checkupdates, it doesn't change anything in the io.pslab.yml file. fdroid lint and fdroid build are successfull.

CloudyPadmal, 2024-05-19:
@AsCress yes we will need to make a version update.

marcnause, 2024-07-01:
After some discussion with @AsCress we noticed that there seems to be an issue with F-Droid not being able to process updates due to missing `versionCode` and `versionName` values in the build.gradle.kts file (they are set via environment variables at build time now) and F-Droid expects the data being present in the build.gradle(.kts) file or in the manifest.

I have read the F-Droid documentation and there may be ways around this limitation, but I am not sure if I understood everything correctly. Therefore I have requested help: https://gitlab.com/fdroid/fdroiddata/-/issues/3300

AsCress, 2024-07-08:
The latest version of the app (3.1.36) is now available on FDroid.

[image]

closed 2024-07-08 by AsCress