fossasia/pslab-app #2428: Wrong plots for the data captured through the In-Built MIC

AsCress, 2024-05-23:
This issue was also noticed by me in the pre-GSoC period, while writing the proposal.
The plot for the waveform captured using the Built-In MIC doesn't seem to be correct. This can be tested by using a tone generator.
**Actual Behaviour**
The captured waveform has different characteristics (time period and frequency) than the one generated by the tone generator.

**Expected Behaviour**
The captured waveform should have the same characteristics the one generated using the tone generator.

**Steps to reproduce it**

1. Turn on the Built-In MIC feature.
2. Use a tone generator to generate any kind of wave (say, sine) of a particular frequency.
3. You will observe that the one shown in the plot has a very different frequency.

**Screenshots of the issue**
Here, is the output for a sinusoidal waveform of frequency 1000 Hz generated using a tone generator.
[image]
The output waveform has a frequency of approx. 20-25 Hz, which is a huge difference from the input waveform.
**Would you like to work on the issue?**
Yes.

AsCress, 2024-05-23:
@marcnause Would you be able to confirm this issue, by trying to reproduce it on your devices ?

AsCress, 2024-05-23:
@CloudyPadmal. Is this supposed to work that way? Or, is it an issue ?

marcnause, 2024-05-23:
> @marcnause Would you be able to confirm this issue, by trying to reproduce it on your devices ?

I tried to reproduce the issue by using the Wave Generator of the PSLab app on one phone and the Oscilloscope screen of the PSLab app on another phone.

The waveform looked similar to the one in the screenshot posted by @AsCress.

I also used the wave generator of the PSLab to generate a sinewave with a frequency of 1000Hz. I connected the Sine Wave 1 output to Channel 1 of the oscilloscope of the PSLab. The result was the waveform in the screenshot below which resembles what I would also expect for the experiment described by @AsCress.

[image]

I guess this also shows that it is not a general problem of the waveform in the Oscilloscope, but it seems to be limited to the In-Build MIC feature. (I haven't tested the MIC input of the PSLab.)

AsCress, 2024-05-25:
@marcnause Thanks so much for confirming this issue! You're correct, as I can see in the code this only seems to be an issue with the MIC and not with any other channels.

PR #2430 (2024-05-26): fix: fixed the plots for the data captured through the In-Built-MIC
Fixes #2428 
This issue was primarily arising due to two reasons:
**1. No manipulation of the xData before plotting -** The xData was not being manipulated before plotting in the case of the In-Built-MIC as it was being done in the other channels. As a result, the data sampled at 44,100 Hz (sampling period approx. 2.26 microseconds) was being plotted at intervals of 1 microsecond, which was causing a waveform of the wrong frequency to be displayed.

**2. Wrong configuration of the millisecond scales -** The xData was not being [... cut]
closed 2024-05-26 by mariobehling