fossasia/pslab-app #2429: Wave Generator crashes on orientation change

marcnause, 2024-05-23:
**Actual Behaviour**

The Wave Generator crashes on orientation change.

**Expected Behaviour**

The app should never crash when the orientation of the device is changed.

1. Open PSLab app
2. Go to Wave Generator
3. Tap "Frequency" button
4. Change frequency via seek bar
5. Change orientation of the device

**LogCat for the issue**
`FATAL EXCEPTION: main
Process: io.pslab, PID: 7816
java.lang.RuntimeException: Unable to start activity ComponentInfo{io.pslab/io.pslab.activity.WaveGeneratorActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.String.hashCode()' on a null object reference
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2678)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2743)
	at android.app.ActivityThread.handleRelaunchActivity(ActivityThread.java:4560)
	at android.app.ActivityThread.-wrap19(ActivityThread.java)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1496)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:154)
	at android.app.ActivityThread.main(ActivityThread.java:6165)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.String.hashCode()' on a null object reference
	at io.pslab.activity.WaveGeneratorActivity.setValue(WaveGeneratorActivity.java:945)
	at io.pslab.activity.WaveGeneratorActivity.-$$Nest$msetValue(WaveGeneratorActivity.java)
	at io.pslab.activity.WaveGeneratorActivity$16.onSeeking(WaveGeneratorActivity.java:432)
	at com.warkiz.widget.IndicatorSeekBar.setSeekListener(IndicatorSeekBar.java:1434)
	at com.warkiz.widget.IndicatorSeekBar.setProgress(IndicatorSeekBar.java:1634)
	at com.warkiz.widget.IndicatorSeekBar.onRestoreInstanceState(IndicatorSeekBar.java:1185)
	at android.view.View.dispatchRestoreInstanceState(View.java:15767)
	at android.view.ViewGroup.dispatchRestoreInstanceState(ViewGroup.java:3240)
	at android.view.ViewGroup.dispatchRestoreInstanceState(ViewGroup.java:3240)
	at android.view.ViewGroup.dispatchRestoreInstanceState(ViewGroup.java:3240)
	at android.view.ViewGroup.dispatchRestoreInstanceState(ViewGroup.java:3240)
	at android.view.ViewGroup.dispatchRestoreInstanceState(ViewGroup.java:3240)
	at android.view.ViewGroup.dispatchRestoreInstanceState(ViewGroup.java:3240)
	at android.view.ViewGroup.dispatchRestoreInstanceState(ViewGroup.java:3240)
	at android.view.View.restoreHierarchyState(View.java:15745)
	at com.android.internal.policy.PhoneWindow.restoreHierarchyState(PhoneWindow.java:2106)
	at android.app.Activity.onRestoreInstanceState(Activity.java:1051)
	at android.app.Activity.performRestoreInstanceState(Activity.java:1006)
	at android.app.Instrumentation.callActivityOnRestoreInstanceState(Instrumentation.java:1196)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2651)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2743) 
	at android.app.ActivityThread.handleRelaunchActivity(ActivityThread.java:4560) 
	at android.app.ActivityThread.-wrap19(ActivityThread.java) 
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1496) 
	at android.os.Handler.dispatchMessage(Handler.java:102) 
	at android.os.Looper.loop(Looper.java:154) 
	at android.app.ActivityThread.main(ActivityThread.java:6165) 
	at java.lang.reflect.Method.invoke(Native Method) 
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:888) 
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:778) 

**Would you like to work on the issue?**
Maybe.

AsCress, 2024-05-25:
@marcnause Shall I work on this one ?

marcnause, 2024-05-25:
I already fixed some problems in https://github.com/marcnause/pslab-android/tree/fixWveGenCrash, but I noticed more problems (data lost on orientation change) which are probably caused by the fact that the screen was designed for fixed orientation only, which was changed later.

If you are looking for something to do, go ahead. You can use or ignore the changes I did in my branch, whatever makes sense to you.

I also noticed more problems in the screen which I would prefer to create an extra issue for to keep the scope of this issue manageable. I just didn't come around to it yet.

AsCress, 2024-05-25:
@marcnause Yeah that's true. There are other issues related to the screen, as you mentioned, including the loss of data when orientation is changed. But, for this issue ticket, I think we should solve only the fact that the "unit" variables becomes null upon orientation change and add a check for that.
We can create new issue tickets for all the other issues and tackle them one by one. Does this sound right ?

PR #2431 (2024-05-25): fix: prevent crash on orientation change in wave generator screen
Fixes #2429

## Changes 
- make sure that `activePropTv` is set on create
- handle `unit` in a `null`-safe way
- use primitive data types instead of objects in a few cases

## Screenshots / Recordings  
N/A

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and  [... cut]
closed 2024-05-25 by mariobehling