fossasia/pslab-app #2432: Wave Generator: layout not really usable in horizontal orientation

marcnause, 2024-05-25:
**Actual Behaviour**

When the Wave Generator screen is used in horizontal orientation, some elements are too small to se or to read the labels on the elements.

**Expected Behaviour**
All UI elements should be easy to use and all labels should be readable. Maybe adding a second layout for horizontal orientation would help.

**Steps to reproduce it**

1. Open app
2. Open Wave Generator screen 
3. Change to horizontal oriientation 

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**
Maybe.

AsCress, 2024-05-26:
@marcnause @CloudyPadmal I see that most of our layouts don't support horizontal orientation, not just this one. Should we keep them as it is to support only vertical orientation ? Or should we create separate layouts for each of the screens to support horizontal orientation ?

marcnause, 2024-05-26:
I requested horizontal layouts some time ago (see https://github.com/fossasia/pslab-android/issues/2119). Since I still own the tablet mentioned in the request, I would still love to have horizontal layouts for every screen.

AsCress, 2024-05-26:
> I requested horizontal layouts some time ago (see #2119). Since I still own the tablet mentioned in the request, I would still love to have horizontal layouts for every screen.

Right, it would be very nice to have horizontal orientation layouts for our screens. I would work on them one by one, starting with the Wave Generator screen, and then we can one by one explore all the other screens (they would also be having several issues on orientation changes as this one).

PR #2439 (2024-05-30): feat: added dedicated horizontal orientation layouts for the Wave Generator screen
Fixes #2432 
Adds dedicated horizontal orientation layouts for the Wave Generator screen, to make it more user-friendly and usable in the horizontal orientation as well, using ScrollViews.

## Changes/Additions
- app/src/main/res/layout-land/activity_wave_generator.xml
- app/src/main/res/layout-land/wave_generator_wave_controls.xml
- app/src/main/res/layout/wave_generator_preview.xml

## Screenshots / Recordings  
A recording of the Wave Generator screen in the horizontal orientation:-

https://github.com/fossas [... cut]
closed 2024-05-30 by mariobehling