fossasia/pslab-app #2442: Wave Generator: seekbar inactive on start of screen even though Freq-button is active

marcnause, 2024-05-28:
**Actual Behaviour**

When the Wave Generator is started, the seekbar at the bottom is inactive, even though the Freq-button is active. The user needs to tap the Freq-button to activate the seekbar.

**Expected Behaviour**

The seekbar should be active and ready to change the frequency since the Freq-button is active.

**Steps to reproduce it**

1. Open app
2. Choose Wave generator

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Maybe

AsCress, 2024-05-29:
@marcnause I guess that the seekbar grayed out is a feature to emphasize that nothing is selected at the start and the user has to click on the property he wants to change. According to me, the frequency button should be greyed out instead at the start. Does that sound right ?

marcnause, 2024-05-29:
Yes, I think that either none of the 3 buttons should be active or the seekbar should be active and set to some default value (e.g. 440 Hz). I am not sure which solution is preferable.

AsCress, 2024-05-30:
According to me, I think we should prefer to keep all the buttons greyed out in the start and let the user select what he wants to set first.

PR #2449 (2024-05-30): fix: fixed the frequency buttons to remain unselected at the start of the wave generator screen
Fixes #2442 
Fixes the frequency buttons to remain greyed out at the start of the wave generator screen as the seekbar is also unselected, indicating that the user would have to select the button (freq/phase/duty) first to set the values.

## Changes 
- app/src/main/res/layout/wave_generator_wave_controls.xml

## Screenshots / Recordings  
Screenshots of the analog and digital wave layouts upon start of the screen:

[image]
[image]

**Checklist**: 
- [x] **No hard coding**: I have used resources fro [... cut]